kad/Marlin #6: anycubic_mega_zero_btt_nano_bed 2.0.7.2 hotbed not reaching target temp

NoobieNoobSquared, 2021-03-09:
New to this. I flashed that firmware.bin to my MZ2.0 with BTT to regain use of hotbed. The issue is when using preheat_PLA or preheat_bed or even using the temp settings menu, it stops a few degrees before reaching target temp and gives an error:
Heating Failed:Bed
Printer Halted
Please Reset

Thank you

kad, 2021-03-11:
yes, 2.0.7.2 had issues with bed config. For MZ2 you need to use dev builds from anycubic-kad-bugfix-2.0.x branch.
binaries can be found [here](http://orava.kad.name/3d/)

ckiefer, 2021-03-12:
Hi Guys. That was very helpful. I had the same issue with the temperature not going over 235 degrees. Using the firmware from the bugfix branch I noticed a couple pixel errors on the display, and also the mesh size is 49 points (instead of 9 as in the stable branch).

kad, 2021-03-12:
> noticed a couple pixel errors on the display
Interesting. which config ? in which area? if possible, can you make a photo of that?

NoobieNoobSquared, 2021-03-12:
Forgive me for my noobness. Maybe someone can point me to a tutorial or link so I can work this. The link has an index with dates. Which date, and which do I select, and how do I install it?

ckiefer, 2021-03-12:
I am using this firmware: amz_melzi_bed_bl_a2-bugfix-2.0.x-20210311-firmware.hex
I didn't have the pixel errors with this one: anycubic_mega_zero_bed_bl_a2-2.0.7.2-firmware.hex

I have added some photos here: https://photos.app.goo.gl/YqFMgkwE3Ro6Ti2Y7

NoobieNoobSquared, 2021-03-12:
I used amz_btt_mini_bed_bl-bugfix-2.0.x-20210312-firmware.bin and I have a working hotbed. Thank you for all the work.

kad, 2021-03-12:
> Forgive me for my noobness. Maybe someone can point me to a tutorial or link so I can work this. The link has an index with dates. Which date, and which do I select, and how do I install it?

@NoobieNoobSquared : 
The latest/ subdirectory contains latest builds. Usually updated daily. If you for some reasons need to revert back to some "known to work" builds, select specific date folder. Inside there are binary files for different boards. How to use them and information about variants are on wiki: https://github.com/kad/Marlin/wiki

kad, 2021-03-12:
@ckiefer can you try to use some older builds, e.g. mid February to see when this effect started to appear? Would be good to bisect on which day in upstream issues being introduced.

ckiefer, 2021-03-12:
@kad Yes, will do that. Is the temperature issue solved in those mid February releases?

kad, 2021-03-12:
@ckiefer MAXTEMP was adjusted in bugfix branch if I remember correctly in end of Dec. MZ2 bed PID parameters commit seems to be on 27th of Dec. Bed preheat PLA menu adjusted on 15th of Feb. And practically since 21st of Feb there were no config changes, only upstream rebases.

ckiefer, 2021-03-13:
@kad The display issues start with this release: amz_melzi_bed_bl_a2-bugfix-2.0.x-20210311-firmware.hex
Before that, it's ok.

kad, 2021-03-13:
> @kad The display issues start with this release: amz_melzi_bed_bl_a2-bugfix-2.0.x-20210311-firmware.hex
> Before that, it's ok.

Thanks for testing @ckiefer !
Moved "latest" symlink to 20210310... so other people will not hit that for now.
It is interesting, as the issue that I suspect was merged before 10th... 
Currently in upstream I don't see neither relevant pull requests nor issue like that reported.
Need to look in more details on that. Can I ask you to make this screen issue as separate issue? so this one can be closed and we continue in new one?

closed 2021-03-16 by NoobieNoobSquared

ckiefer, 2021-04-07:
Hi @kad
This link: http://orava.kad.name/3d/ is not working anymore. Did you move the files?

kad, 2021-04-07:
> Hi @kad
> This link: http://orava.kad.name/3d/ is not working anymore. Did you move the files?

it was issue with dynamic DNS not updated properly. please re-check, should be accessible now.