kad/Marlin #9: Enable both PIN27 and zmin for bltouch and PIN29 for filament sensor

pappicio, 2021-03-25:
Hi, I tryed to enable both pin27 and pin29, firts for bltouch with zmin stop, and also pin29 for filament sensor, result: filament sensor on pin29 doesn't work.
I know, is not implemented into your releases, but adding _fs_ on one of choice to compile, visual studio code add all needed, and also menu in the printer screen shows rounout (on/off), there is a possivility to add them both working?

kad, 2021-04-03:
In anycubic-kad-bugfix-2.0.x branch, anycubic_mega_zero_melzi_bl_zmin_fs or anycubic_mega_zero2_melzi_bl_zmin_fs should cover that config. btw, sensor by default is not activated, check in configuration menu to enable it, that also could be reason. (or see https://marlinfw.org/docs/gcode/M412.html to enable that over terminal).

pappicio, 2021-04-04:
Hi, thanks, I have mega zero 1.0 and also heated bed, so the firt choice, is bad for me, but, can I use instead some mega zero 2.0 build, like "nycubic_mega_zero2_melzi_bl_zmin_fs", cos it has iet actived heat bed?
thanks.

kad, 2021-04-04:
@pappicio I've uploaded to http://orava.kad.name/3d/experimental/ 

amz_melzi_bed_bl_zmin_fs-bugfix-2.0.x-20210403-firmware.hex
amz_melzi_bed_bl_zmin_sfs-bugfix-2.0.x-20210403-firmware.hex

please try (sfs for BTTs smart filament sensor, fs for simple/cheap ones).
If those are ok for you, I'll add those to regularly generated set.

pappicio, 2021-04-04:
waw, finally I can compile with all upgrades I have, without suppress sdcard support. For now just compiled, If I have some minutes free tomorrow I'll try to use filament sensor (I use simple, with only 2 wires: first connected on PIN29, second on GND). I'll tell you if all works with it, asap! thanks.

pappicio, 2021-04-05:
ok.tested amz_melzi_bed_bl_zmin_fs-bugfix-2.0.x-20210403-firmware.hex.all works except runout filament (I repeat simple filament sensor on EXT - A2, 2 wire: pin 29 and GND). gcode M503, about tell me no info (no triggered or open info on M503 log...)
On lcd I only can see enable filament sensor; connecting sensor on melzi board, nothing appen during rinting; runout filament connected on rasberry: it only stops printing cos send pause print to octoprint, but nozzle stay there on printing area.
If I enable also: avanced pause and nozzle park, if open filament sensor on raspberry, it pause print, goes safe area of printing area and wait for filament change, and on lcd I can follow instrucyions about; at filament changed, I can confirm, nozzle pre heat and then print resume (only after I pressed also button "resume" on octoptint interface), so:
If I disable SDCARD SUPPORT and force to compile also avanced pause and nozzle park, having more free flash ROM, amd only with support of octoprint (send M600 command to the printer after it senses print pause by its own filament sensor),  I'm able to change filament in safety mode and can then resume print.
I only hope that will be possible without octoprint after I've received new MINI MZ board for my Mega Zero Printer.
btw, thanks a lot.

kad, 2021-04-05:
On Melzi board, there is no really space for M600, that's why it's disabled. On Melzi only possibility is to do M25 (sd print pause), thus functionality is limited to SD printing only.
For M119: if you don't see changes in state if you're inserting or removing filament - recheck wiring. pull-up resistor on that input is enabled, so if it correctly shortened to ground, it should be in low state.

pappicio, 2021-04-05:
yes, thanks for confirmation, I tested with a multimeter, on the filament sensor, the 2 wires (PIN29 and GND) goes to ground when connected, (I can see 5v, if I remember well) but nothing appen if I open contact by switch (so, PIN 29 goes to pullup), print doesn' t pause. I think maybe something is wrong... but here, someone else who tryes filament sensor, could confirm or not.

pappicio, 2021-04-05:
okok, something seems to move now: ok, I use Marlin to remotely print, not need use builtin SDCARD, so:
in "Configuration.h" I modified:
this
#if ENABLED(KAD_MELZI)
    #define FILAMENT_RUNOUT_SCRIPT "M25"
to
#if ENABLED(KAD_MELZI)
    #define FILAMENT_RUNOUT_SCRIPT "M600"
this
// #if ENABLED(KAD_MELZI) && ANY(KAD_SMART_FILAMENT_SENSOR, KAD_FILAMENT_SENSOR) && DISABLED(KAD_BLTOUCH)
//   #define NOZZLE_PARK_FEATURE
to
#if ENABLED(KAD_MELZI) && ANY(KAD_SMART_FILAMENT_SENSOR, KAD_FILAMENT_SENSOR) /////  && DISABLED(KAD_BLTOUCH)
   #define NOZZLE_PARK_FEATURE
this
#define SDSUPPORT
to
#if ENABLED(KAD_MELZI) && !ANY(KAD_SMART_FILAMENT_SENSOR, KAD_FILAMENT_SENSOR, KAD_BLTOUCH)
   #define SDSUPPORT

(so disable DS CARD SUPPORT to have many more flash ROM feee)

in "Confoguration_adv.h:
this
// #if ENABLED(KAD_MELZI) && ANY(KAD_SMART_FILAMENT_SENSOR, KAD_FILAMENT_SENSOR) && DISABLED(KAD_BLTOUCH)
//   #define ADVANCED_PAUSE_FEATURE
to
#if ENABLED(KAD_MELZI) && ANY(KAD_SMART_FILAMENT_SENSOR, KAD_FILAMENT_SENSOR) ///// && DISABLED(KAD_BLTOUCH)
   #define ADVANCED_PAUSE_FEATURE

in my way I have: Heated bed (with added mosfet); bltouch (on PIN27 and Z stop); Flament Sensor (ON EXT-A2, PIN29-GND)

I enabled ADVANCED_PAUSE_FEATURE and NOZZLE_PARK_FEATURE, disabled SDSUPPORT, and enabled M600 Instead of M25.
Now, during print, if I remove filament from 2 wire Filament Sensor (installed on PIN29/GND), after 5mm of filament (I setted limit to 5mm...), print pause, nozzle goes to secure position and lcd tell me to change fillement, anfer followed instruction appeared on lcd, I can resume print, and without do anyting on octoptint interface, I'm happy (shure, I've lost SD CARD, but really noe need it, I watch tv from my livingroom and choice the object to print by octoprint interface!)
Thanks for your time, and in future, if you want make this modification to add full support for filament sensor joined to bed and bltouch, surelly someone else will be appy about!
see you!

kad, 2021-04-06:
ok. I'll keep that issue then in that state, if someone wants to repeat it.
I think loosing SD support might be ok for some, but that probably needs to be conscious decision :)
Generally, it is anyway better to upgrade from stock Melzi board to something else (at least, for the sake of better motor drivers).
Thx for trials.

closed 2021-04-06 by kad

pappicio, 2021-04-06:
yes I know, is better change board, in fact I'm waiting for SKR mini mz (specific board for mega zero 1.0), but while wait,  I like to have all upgrades ready and working, and now I have all them. yes, only disabling sdcard.
btw, thanks for all! maybe I'll return here for new questions when I'll have under hands new board! :)