kad/Marlin #10: fan

velidhu, 2021-03-27:
can you tell me how to enable the extruder fan which starts working when the extruder temperature reaches 50 °

kad, 2021-04-03:
You need to have firmware that is marked with e0fan in the filename. That would control fan labeled on SKR boards as CNC_FAN2 as automatic extruder fan.

ckiefer, 2021-04-08:
@kad That is helpful. 
On my SKR mini E3 V2.0 board I see locations 'FAN0' and 'FAN1', but I don't see a spot labelled 'CNC_FAN2'.

kad, 2021-04-08:
@ckiefer yes, on E3 fans are done differently, thus SKR mini E3 has different labels.
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V2.0/Hardware/BTT%20SKR%20MINI%20E3%20V2.0-PIN.pdf 
FAN0 (PC6) is for part cooling, FAN1 (PC7) by default Controller Fan or in `e0fan` - extruder automatic fan.

closed 2021-06-12 by kad