kad/Marlin #13: Bigtreetech SKR Mini E3 V2.0 Mainboard Installation (Findings and Issues)

ckiefer, 2021-04-17:
I finally managed to install my new BTT SKR Mini E3 V2.0 board in my Anycubic MZ 2.0. The dimensions of the board do not precisely fit the Anycubic MZ mainboard/display case so I had to fiddle a bit ... Now it's connected with a single screw and is holded by the USB connector coming from my Raspberry Pi on the other side. Not perfect but it works. If you asked me why I didn't install the BTT SKR Mini MZ V1.0 from the beginning, then it's because I wrongly ordered it because I was not aware of that fact that the board also exists for the MZ ...

I have installed the BLtouch to the Z-Probe connector and the hotend fan to fan1 connector. I have also installed a filament runout sensor and connected it to E0-Stop connector. Furthermore I've installed an 8-LED strip and connected it to the Neopixels port.

I have also connected a BTT 3.5" TFT display. Here I connected EXP3 on the display to EXP1 on the board and the TFT connector on the board to the respective connector at the bottom on the display.

This document helped me to do the correct wiring:
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V2.0/Hardware/BTT%20SKR%20MINI%20E3%20V2.0-PIN.pdf

I also installed an BTT ESP-01S wifi module following this guide:
https://youtu.be/AGTF-JEUHNA

Also this guide was helpful:
https://youtu.be/mtCz_-2zvZo

Another large problem was the power (5V) coming from the USB cable that connected my Raspberry Pi to the BTT SKR Mini E3 V2.0 board. There are happening all sorts of errors if both power from the Pi and the printer's main PSU comes in. I therefore had to manually cut the red wire inside the USB cable to disable powering the board. The stock melzi board that came with the Anycubic MZ 2.0 printer "couldn't use" the power from the Pi. A whole thread about this can be found here:
https://community.octoprint.org/t/put-tape-on-the-5v-pin-why-and-how/13574

That would also be nice solution (untested) to solve this eventually:
https://www.thingiverse.com/thing:3044586

To install the firmware I performed these steps:
1. Checked out Github branch anycubic-kad-bugfix-2.0.x
2. Installed VSCode with the PlatformIO and Auto Build Marlin plugins
3. Opened the checked-out folder in VSCode
4. Went to Project Tasks under the PlatformIO plugin
5. Build the target 'anycubic_mega_zero2_btt_mini_bl_e0fan
6. Copied the generated firmware as firmware.bin to an SD card
7. Inserted the SD card into the mainboard and powered-up the printer to flash the firmware

This guide was helpful:
https://youtu.be/eq_ygvHF29I

Here are some findings:
- **Issue**: Although I set to print the filament with 255 degrees (hotend temperature), the hotend only heats up to 245 degrees. Don't know how to change this. Here's a snippet of the GCode:

; generated by SuperSlicer 2.3.55 on 2021-04-15 at 08:00:54 UTC

; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.45mm
; first layer extrusion width = 0.45mm

; object:{"name":"RaspberryPi_Camera_Cover_Back","id":"RaspberryPi_Camera_Cover_Back.STL id:0 copy 0","object_center":[74.665106,142.184334,0.000000],"boundingbox_center":[74.665106,142.184334,1.500000],"boundingbox_size":[28.000000,28.000000,3.000000]}
; object:{"name":"RaspberryPi_Camera_Cover_Front","id":"RaspberryPi_Camera_Cover_Front.STL id:1 copy 0","object_center":[144.743497,69.358194,0.000000],"boundingbox_center":[144.743497,69.358194,4.000000],"boundingbox_size":[28.000000,46.000000,8.000000]}
; Total objects to print: 2
M486 T2
; plater:{"center":[109.704302,101.271264,0.000000],"boundingbox_center":[109.704302,101.271264,4.000000],"boundingbox_size":[98.078391,109.826139,8.000000]}

M140 S112 ; set bed temperature
;TYPE:Custom
G28 ; home all axes
M155 S30
@BEDLEVELVISUALIZER
G29 T ; BL Touch mesh generation
M155 S3
G1 Z5 F5000 ; lift nozzle
M107
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
; Filament gcode
M109 S255 ; set temperature and wait for it to be reached
M190 S112 ; set bed temperature and wait for it to be reached
...

- **Issue?**: The LED are shortly red at the beginning of the print, then slightly change to violetish, and then are always green while printing (not sure if that's supposed to be like this). I can change the LED color through the display. I am not sure if this is an issue. Basically I don't know how these LEDs are supposed to work anyways ...

Suprisingly, the fact that I cannot print at 255 degrees doesn't play a big role at the moment. I am mostly printing with ABS and those prints seem to work with 245 as well.

I've attached a couple of pictures of my setup. @kad Really nice work on this webpage, thx a lot!

[image]
[image]
[image]

kad, 2021-04-17:
Regarding USB power: I luckily had USB "pigtails" with male/female connectors, so I just made "adapter" cable with GND/D+/D- connected only.

245 degrees: that is intentional. 245ºC is the melting point of PTFE tube, and as MZ uses "bore" throat, it means PTFE tube will be exposed pretty much to the maximum temp of it. It is configured in https://github.com/kad/Marlin/blob/anycubic-kad-bugfix-2.0.x/Marlin/Configuration.h#L521-L543 So, if you changed to e.g. "all metal" type of throat or extruder, you can update maxtemp values (note about `overshoot` constants).

Regarding builds: I'm often publish dev builds on my dev host: http://orava.kad.name/3d/ to avoid people to need to use vscode. But it is of course nice when people are able to work with source code, so there is a hope on contributions in terms of PRs :)

ABS: it is generally not best idea to print ABS without chamber or at least good ventilation. +note about PTFE tube in hotend. So, for ABS I think it is anyway falls to "advanced" usage of MZs.

Side note about Slicer output: you don't need to do G29 every print. It is enough to get leveling done, store it to eeprom, and then in start gcode to have "m420 s1" to read eeprom mesh data into memory. re-leveling can be done once in few weeks, or if something changes with bed (e.g. using different surfaces from time to time like glass/pei metal sheets).

Neopixel leds: it is described here: https://github.com/kad/Marlin/blob/anycubic-kad-bugfix-2.0.x/Marlin/Configuration.h#L2905-L2918 However, if print is done via OctoPrint or from BTT TFT media, then it might not work as in defaults. check settings in TFT (as it can control leds) or add your custom G-codes to end-script in slicer.

ckiefer, 2021-04-18:
@kad Thanks for the useful hints.

So then I guess the points above are more for discussion than to be treated as issues. 

I have upgraded my printer with Capricorn PTFE tubes a while ago. I'm by far not an expert but these resources say that 255 is fine for PTFE that is in direct contact with the nozzle / hotend:
- https://www.captubes.com/safety.html
- https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-general-discussion-announcements-and-releases/ptfe-max-temp-on-a-routine-basis/
- http://www.rjchase.com/ptfe_handbook.pdf

How would I change the temperature settings?

Regading ABS, I will later attach my SuperSlicer profiles for ABS which work well for my Anycubic MZ.

Regarding Neopixels, I would change this line
#define NEOPIXEL_PIXELS 16 into #define NEOPIXEL_PIXELS 8 becaue I only have 8 LEDs, right?

I could also uncomment line //#define TEMP_STAT_LEDS, right?

You said "check settings in TFT (as it can control leds) or add your custom G-codes to end-script in slicer". Could you elaborate a bit more on that? What type of settings in the TFT and could you post an example Gcode for the slicer to give me an idea?

kad, 2021-04-18:
Well, if you're sure about your replaced tube, then adjust `HEATER_0_MAXTEMP ` https://github.com/kad/Marlin/blob/anycubic-kad-bugfix-2.0.x/Marlin/Configuration.h#L524 to the desired max temp + `HOTEND_OVERSHOOT` and rebuild. E.g. 255+15 = 270.

`NEOPIXEL_PIXELS` - doesn't really matter 16 or 8. Only if you want to enable NEOPIXEL_IS_SEQUENTIAL, otherwise it is anyway changing all pixels at once, so amount of them doesn't matter.

`TEMP_STAT_LEDS` - if you like that function of LEDs, yeah, why not :)

G-code control for neopixels: https://marlinfw.org/docs/gcode/M150.html
With `I` parameter you can set individual pixel color, and without it - for whole stripe.
So, in the stop gcode you can add something like `M150 U255 B0 R0 W0 P127` to make them green.

ckiefer, 2021-04-18:
Thanks @kad. That was helpful. Using `#define HEATER_0_MAXTEMP 270` I can heat up the nozzle again to 255 degrees.

For some, reason whatsoever, my TFT now does not update the nozzle/bed temperatures anymore. During heating up, the values freeze at 64 and 38 (see picture). No clue ...

[image]

I uncommented `//#define TEMP_STAT_LEDS` in Configuration.h and received this error:

`
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\HAL.cpp:30:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:1974:4: error: #error "TEMP_STAT_LEDS requires STAT_LED_RED_PIN or STAT_LED_BLUE_PIN, preferably both."
`

How can I find out the pins for `STAT_LED_RED_PIN` AND `STAT_LED_BLUE_PIN`?

For the moment, I use these settings:

```
//#define TEMP_STAT_LEDS

//#define RGB_LED
//#define RGBW_LED

#if EITHER(RGB_LED, RGBW_LED)
  //#define RGB_LED_R_PIN 34
  //#define RGB_LED_G_PIN 43
  //#define RGB_LED_B_PIN 35
  //#define RGB_LED_W_PIN -1
#endif

#if ENABLED(KAD_SKR_ENOUGH_FLASH)
  #define NEOPIXEL_LED
#endif
#if ENABLED(NEOPIXEL_LED)
  #define NEOPIXEL_TYPE   NEO_GRB  
  #define NEOPIXEL_PIN    PA8   
  //#define NEOPIXEL2_TYPE NEOPIXEL_TYPE
  //#define NEOPIXEL2_PIN    5
  #define NEOPIXEL_PIXELS 8      
  #define NEOPIXEL_IS_SEQUENTIAL  
  #define NEOPIXEL_BRIGHTNESS 127
  #define NEOPIXEL_STARTUP_TEST 

  //#define NEOPIXEL2_SEPARATE
  #if ENABLED(NEOPIXEL2_SEPARATE)
    #define NEOPIXEL2_PIXELS      15
    #define NEOPIXEL2_BRIGHTNESS 127
    #define NEOPIXEL2_STARTUP_TEST
  #else
    //#define NEOPIXEL2_INSERIES
  #endif

   //#define NEOPIXEL_BKGD_LED_INDEX  0
  //#define NEOPIXEL_BKGD_COLOR { 255, 255, 255, 0 }
  //#define NEOPIXEL_BKGD_ALWAYS_ON
#endif

#if ANY(BLINKM, RGB_LED, RGBW_LED, PCA9632, PCA9533, NEOPIXEL_LED)
  #define PRINTER_EVENT_LEDS
#endif
```

kad, 2021-04-19:
TFT temp: try to open terminal and see if it reports temperatures there. it should be regularly send temp status updates.
`TEMP_STAT_LEDS` - I haven't checked those myself, need to look. will comment once will look at it.

closed 2021-06-12 by kad