kad/Marlin #14: SKR MINI (MZ) Board and simple filament sensor

pappicio, 2021-04-23:
Hi, I've found that SKR releases only uses smart filament sensor, infact, If I compile and use SKR_BTT_MINI (for mega zeor 1.0 version) and enable filament_sensor, I have false positive, so when start rpinting, printer tell ever me that filamen is terminated, so I changed:
    #if ANY(KAD_SKR_ENOUGH_FLASH, KAD_SMART_FILAMENT_SENSOR)
       #define FILAMENT_MOTION_SENSOR
    #endif
to:
    #if ANY(KAD_SKR_ENOUGH_FLASH, KAD_SMART_FILAMENT_SENSOR)
    ///  #define FILAMENT_MOTION_SENSOR
    #endif

and finally it worked with simple filament sensor, in fact setting simple filament sensor pin to GND, and only in this case I receive on lcd screen message that filament is terminated and need to be changed.
so at end I modified to:
///// #if ANY(KAD_SKR_ENOUGH_FLASH, KAD_SMART_FILAMENT_SENSOR) //this line
    #if ENABLED (KAD_SMART_FILAMENT_SENSOR) //with this one
      #define FILAMENT_MOTION_SENSOR
    #endif
 so, ony if I defined smart filamet sensor,   #define FILAMENT_MOTION_SENSOR is Enabled and now works well, if I have simple filament on octoprint raspberry3 pins, and prefer use it to have also messages sent to homeassistant and then adviced by alexas diposed into my home, but, in this way, simple filament sensor finally works.
thanks.

kad, 2021-04-25:
Yes, BTTs smart filament sensor is better choice than simple ones, so for SKR boards it is selected as default option.
So far nobody requested other configuration, so I haven't added it as separate build option, and it is easy to comment out one line if needed. If there will be need to persistently generate such configuration, I can add build targets for that as well.

pappicio, 2021-04-26:
ok. no problem, disabling that line works well, for me is ok!

pappicio, 2021-05-08:
I can see yoy 're updating skr mini buils inserting also fs for filament sensor!
Ive added also theese ones!

[env:anycubic_mega_zero_btt_mini_bl_zmin_e0fan_fs]
extends                     = env:anycubic_mega_zero_btt_mini
[env:anycubic_mega_zero_btt_mini_bed_bl_fs]
extends                     = env:anycubic_mega_zero_btt_mini
[env:anycubic_mega_zero_btt_mini_bed_bl_zmin_fs]
extends                     = env:anycubic_mega_zero_btt_mini
[env:anycubic_mega_zero_btt_mini_bed_bl_e0fan_fs]
extends                     = env:anycubic_mega_zero_btt_mini
[env:anycubic_mega_zero_btt_mini_bed_bl_zmin_e0fan_fs]
extends                     = env:anycubic_mega_zero_btt_mini
[env:anycubic_mega_zero_btt_mini_bfpt_fs]
extends                     = env:anycubic_mega_zero_btt_mini
[env:anycubic_mega_zero_btt_mini_bfpt_e0fan_fs]
extends                     = env:anycubic_mega_zero_btt_mini
[env:anycubic_mega_zero_btt_mini_bfpt_zmin_fs]
extends                     = env:anycubic_mega_zero_btt_mini
[env:anycubic_mega_zero_btt_mini_bfpt_zmin_e0fan_fs]
extends                     = env:anycubic_mega_zero_btt_mini
[env:anycubic_mega_zero_btt_mini_pinda_fs]
extends                     = env:anycubic_mega_zero_btt_mini
[env:anycubic_mega_zero_btt_mini_pinda_e0fan_fs]
extends                     = env:anycubic_mega_zero_btt_mini
[env:anycubic_mega_zero_btt_mini_bed_bfpt_fs]
extends                     = env:anycubic_mega_zero_btt_mini
[env:anycubic_mega_zero_btt_mini_bed_bfpt_zmin_fs]
extends                     = env:anycubic_mega_zero_btt_mini
[env:anycubic_mega_zero_btt_mini_bed_bfpt_e0fan_fs]
extends                     = env:anycubic_mega_zero_btt_mini
[env:anycubic_mega_zero_btt_mini_bed_bfpt_zmin_e0fan_fs]
extends                     = env:anycubic_mega_zero_btt_mini
[env:anycubic_mega_zero_btt_mini_bed_pinda_fs]
extends                     = env:anycubic_mega_zero_btt_mini
[env:anycubic_mega_zero_btt_mini_bed_pinda_e0fan_fs]
extends                     = env:anycubic_mega_zero_btt_mini

kad, 2021-05-08:
@pappicio yeah, I'll need to fix script that generates all those permutations of configs and update it globally. I had in recent few days person who is trying to use simple filament sensor with BTT SKR mini board, thus adding few configs for him to test.
One feedback that I've got is that for his sensor it would be needed to invert FIL_RUNOUT_STATE=LOW to FIL_RUNOUT_STATE=HIGH. @pappicio how is it in your setup?

pappicio, 2021-05-08:
I use: mini_bed_bl_zmin_e0fan_fs and works well, and nothing modded respect original code, so It is (if I remeber well) pulled up and FIL_RUNOUT_STATE=LOW  (If It means that filament is not present). I used homemade sensor 3d printed and a switch, so I've connected only 2 wire. Pin-in and GND, so when filament ends, the switch close and Pin goes to gnd.

kad, 2021-05-08:
reprap link is not working for me :(
ok, anyway, it seems I need to connect and re-check with sensors I have.

pappicio, 2021-05-08:
[image]
you can see It now?

kad, 2021-05-08:
looks similar to what I bought from aliexpress, but not sure about presence of R1... anyway, I'll need to try myself it seems.

pappicio, 2021-05-08:
R1 is to put pin in pullup with 5v but is bypassed by led and 1k resistor that are from Pin to 5v so led in on and only when switch turns to GND, the resistor R1 became scrive for a while, leaving Pin to pullup, until  itself is connected to GND, in trust case passes from HIGH state to LOW state (filament ended!)

kad, 2021-05-08:
@pappicio I understand that :) I'm just not sure the ones I have have similar schematics, need to double check those.

pappicio, 2021-05-08:
ok, btw,  thanks for all your work on marlin firmware.

pappicio, 2021-05-09:
just 10 minutes agoo I contacted a friend of mine, he has simple filament sensor and just for wiring, he confirm, 3 wire and the sequence is: +/-/in (he told me about the wires colors: red/black/white)...

pappicio, 2021-05-09:
well, I've seen on thingiverse a model of runout sensor for ender 3, with schematic, and it had wire inverted in 3.3/5v and GND, it had: GND/5V/signal.... very strange.... maybe depend only by the sensor you bought. (if I find again that page I'll post schematic, that, repeat, is inverter from that one I've posted.....)

kad, 2021-05-09:
this one is DIY variant... my point was what to keep as defaults in my configs. Based on what I was able to buy - LOW is the setting. For more rare variants people might need to update themselves.

pinout above: this one above targeting to be used with some pin27 adapters on original Melzi boards. Those have different pinouts, e.g. easier BLTouch 3pin servo connection. so, orders of the wires on DIY variants not that matter...

Reversing 5v/gnd might work, need to re-check at least in mine, how the LEDs are wired. I think in cheaper versions from Aliexpress they are saving pennies on resistors/capacitors... so will not be surprised if that will not work :)

pappicio, 2021-05-26:
some news (I had runout filament sensor on raspberry3 before) now I changed the raspberry with a minipc, rellay mini, but powerfull respect to raspberry, so, I now mounted rounout filament sensor on SKR mini MZ (not previous homemade 2 wire filament sensor, but new one with 3 wire by aliexpress), and, yes, I can confirm that also for my 3 wire rounout filament sensor I need to change:

FIL_RUNOUT_STATE=LOW 
to 
FIL_RUNOUT_STATE=HIGH

so: homemade  2 wire runout filament sensor, source is ok.
3 wire filament sensor I need to change, and then all works well (I re-fixed to work property also advices by Homeassistant, telegram and alexa announces, if filamet ends during print operations!)

kad, 2021-05-26:
@pappicio it looks like there are several designs. Mine are from older designs where state is LOW to trigger. One of the vendors nowadays changed description to state=HIGH. So, they might have newer variant.

Regarding reversing 5v/gnd: I tried that. Yes, it also inverts the signal pin, however, that disables LED on sensors (well, polarity of LED is becoming opposite, as expected).