kad/Marlin #18: Sensorless SKR Mini MZ "Y" Axis

DANELSC, 2021-06-05:
Hi, I have a problem with the activation of the sensorless in the axis y, I can't get it to work in this firmware, nor adjusting the sensitivity to the maximum, if I have achieved it in the Marlin of Bigtreetech, but I love this work of KAD, do I  can you give any idea?  Thank you so much.

kad, 2021-06-06:
Which lines you changed in Configuration.h / Configuration_adv.h ? Also, did you put jumpers on the board for sensorless homing?
I haven't tried it myself (prefer hardware endstops rather than crashing carriages on the frame), but as I recall in configs, you can't enable it for just single axis, but I might be wrong.

DANELSC, 2021-06-07:
> Which lines you changed in Configuration.h / Configuration_adv.h ? Also, did you put jumpers on the board for sensorless homing?
> I haven't tried it myself (prefer hardware endstops rather than [...]

Hello Mr KAD, very grateful for your support.
 I have adjusted the values ​​that I indicate in the attached image, I have also added to the board its corresponding Jumper in the X and Y axis. With this same configuration I have achieved satisfactory results to work in sensorless in the Bigtreetech firmware for SKR Mini MZ.

 I am not an expert in programming, just an apprentice in 3d printing.

 Thank you very much for your help and work.
[image]