kad/Marlin #19: Octoprint Action Commands

GimmiT, 2021-06-06:
Is it posible to change. Uncomment #define HOST_ACTION_COMMANDS in Configuration_adv.h . This would allow for better integration with octoprint. Or would this lead to other consequences? I'm currently running: amz_btt_mini_bed_bl-2.0.8.2-kad1-firmware.
source:
https://community.octoprint.org/t/octoprint-tells-me-my-firmware-lacks-support-for-host-action-commands-what-does-this-mean/34588

pappicio, 2021-06-06:
I've compiled amz_btt_mini_bed_bl_zmin_e9fan_fs with uncommented #define HOST_ACTION_COMMANDS,
and all works well, nothing strange to advice about...

kad, 2021-06-06:
Noted, will check if I can fit those on Melzi, but for BTT boards will enable.

GimmiT, 2021-06-07:
Thank you. Will this be in a new version or will you update the old ones.

kad, 2021-06-07:
@GimmiT not sure yet. I was planning to enable that first in bugfix-2.0.x, but it is currently broken for SKR boards due to migration to stm32 HAL, and I see activity in 2.0.x branch, so might be 2.0.8.3 will be released soon.

kad, 2021-06-13:
@GimmiT bugfix-2.0.x-20210613 and up would have that option enabled in SKR configs. For stable builds - if it will be updates to 2.0.8.x or newer - will also enable similarly.

closed 2021-06-13 by kad