kad/Marlin #20: Build Varients Wiki

Vicarious111, 2021-06-06:
The build variants wiki is not using the current build abbreviations.  While they can still be (mostly) figured out, they can be confusing.

Kirottu, 2021-06-29:
Yeah i have noticed, I for one would like to get manual mesh bed leveling on my original AMZ but I don't know which one of them is the right one as the naming scheme has been changed.

Bischi1997, 2021-07-13:
I asked in the AMZ sub. The right version with manual bed leveling for the AMZ 1.0 with the original melzi board version should be the following: amz_melzi-2.0.9.1-kad1-firmware.hex

If I understood correctly the abbreviations stand for the following:
akp -> allen key probe
bl -> BLTouch
bfpt -> BFPTouch
PINDA  -> PINDA sensor

The other abbreviations are explained in the [Leveling Sensors Wiki](https://github.com/kad/Marlin/wiki/Leveling-sensors)

If I missed something, let me know :)

kad, 2022-01-07:
Build variants page is updated. If there are comments or suggestions, please speakup :)

closed 2022-01-07 by kad