kad/Marlin #21: BLTouch Issue

ckiefer, 2021-06-14:
Hello All

I downloaded and built the firmware from branch anycubic-kad-bugfix-2.0.x and applied the following changes to fit my needs:

**configuration.h**

#define GRID_MAX_POINTS_X 3

#define NEOPIXEL_TYPE   NEO_GRB  // NEO_GRBW / NEO_GRB - four/three channel driver type (defined in Adafruit_NeoPixel.h)
#define NEOPIXEL_PIN     PA8     // LED driving pin
//#define NEOPIXEL2_TYPE NEOPIXEL_TYPE
//#define NEOPIXEL2_PIN    5
#define NEOPIXEL_PIXELS 8       // Number of LEDs in the strip. (Longest strip when NEOPIXEL2_SEPARATE is disabled.)
#define NEOPIXEL_IS_SEQUENTIAL   // Sequential display for temperature change - LED by LED. Disable to change all LEDs at once.
#define NEOPIXEL_BRIGHTNESS 255  // Initial brightness (0-255)
#define NEOPIXEL_STARTUP_TEST

#define HEATER_0_MAXTEMP 270

#define Z_PROBE_FEEDRATE_FAST (200*60)
#define MULTIPLE_PROBING 2

**configuration_adv.h**

#define HOST_ACTION_COMMANDS
#if ENABLED(HOST_ACTION_COMMANDS)
  #define HOST_PROMPT_SUPPORT
  #define HOST_START_MENU_ITEM  // Add a menu item that tells the host to start
#endif

Not sure why but I keep getting this error:

[image]

The BLTouch keeps blinking and I am not able to connect anymore in OctoPrint.

Any help is highly appreciated.
Christoph

GimmiT, 2021-06-16:
Are you sure this couldn't be a hardware problem. Had a similar problem today where my BLTouch was not properly connected to the motherboard on only 1 pin. Didn't look like it doesn't work, but it crashes as soon as it tries to z-Home.

ckiefer, 2021-06-17:
@GimmiT Yes, it only happens when z home the printer. The connection to the board looks good, I checked that. I realized that I still had the z-min endstop connected to the board. Maybe that's the reason? Do I have to unplug the z-min endstop?

GimmiT, 2021-06-17:
No zmin shouldn't be a problem as I have it stil installed and everything works. I don't know what it exactly could be. Try deactvating z_Probefeedrate and Multiple_Probing. For Octoprint I personally recommend the bed visualizer with the following commands M28 M29 and M500 this will level your bed and save the data on your printer even in a case of turning the printer off. This should eliminate the need for fast and double probing before every print. Please remember to probe at the right bed temperature.

kad, 2021-06-24:
Can you post photo on how your wiring of bltouch connected to the board?
Also, output in octoprint "terminal" tab might help with decrypting what is going on with error.

Other note: redefining `NEOPIXEL_PIN` is not needed if you're using standard port on SKR mini board.

ckiefer, 2021-07-19:
Hi @kad and @GimmiT 
Thanks for the valuable insights.
It turned out that the cable between the BLTouch and the mainboard was defective. I ordered a new sensor and it came with a new cable. I replaced the old sensor (because I also broke it when I tried to repair the cable and everything ended up in an unattractive grey smoke ...). Since then the error did not show up again.

GimmiT, 2021-07-19:
Thank you. Had to replace my cable too. If you haven't allready I highly recommend taking a cable that is not Extended with plugs but 1 cable only. The cable I currently use is this: https://www.3dware.ch/BLTouch-Servo-Verl%C3%A4ngerungskabel-Set-1000mm-De.htm .It removes most potential for lose wires for me.