kad/Marlin #27: Linking Fails for anycubic_mega_zero2_btt_mini_bl_e0fan

ckiefer, 2021-07-14:
Target 'anycubic_mega_zero2_btt_mini_bl_e0fan' fails to build in VSCode after I make the following changes:

**configuration.h:**

#define GRID_MAX_POINTS_X 3

#define NEOPIXEL_PIXELS 8
#define NEOPIXEL_IS_SEQUENTIAL  
#define NEOPIXEL_BRIGHTNESS 255
#define NEOPIXEL_STARTUP_TEST

#define HEATER_0_MAXTEMP 270

**configuration_adv.h**

#define HOST_ACTION_COMMANDS
#if ENABLED(HOST_ACTION_COMMANDS)
  #define HOST_PROMPT_SUPPORT
  #define HOST_START_MENU_ITEM
#endif

This is the end of the error message in the terminal:
[image]

ckiefer, 2021-07-19:
In the meantime I found out that disabling again HOST_PROMPT_SUPPORT and HOST_START_MENU_ITEM allows me to build (i.e., link) the firmware again.

#if ENABLED(KAD_SKR_ENOUGH_FLASH)
  #define HOST_ACTION_COMMANDS
#endif
#if ENABLED(HOST_ACTION_COMMANDS)
  //#define HOST_PROMPT_SUPPORT
  //#define HOST_START_MENU_ITEM  // Add a menu item that tells the host to start
#endif

There is not enough flash size for this on the mainboard?

kad, 2021-08-22:
Yes, many of the SKR mini environments are close to the flash size limit. if enabling something in addition, it is good to disable something else. host prompts are good candidates for that.

closed 2021-08-22 by kad