kad/Marlin #30: Strange PSU Click Noise

ckiefer, 2021-07-19:
Hello All

For some reason whatsoever my PSU makes strange click noises:
https://photos.app.goo.gl/tWFLZF5hVtahtbde6

I goes hand in hand with the light on the mainboard that blinks whenever the temperature for the heatbed should be kept at (in my case) 60 degrees. It started only some weeks ago and was not there before.

I am building target anycubic_mega_zero2_btt_min_bl_e0fan.

Any ideas how I can get rid of those click sounds again?

Cheers, Christoph

BTW: In the video you can also see a couple of buck converters that are needed to use my 12V Noctua fans.

GimmiT, 2021-07-28:
What you are experiencing most likely is a spike in current that is drawn. https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-hardware-firmware-and-software-help/mk3-power-supply-clicking/

ckiefer, 2021-07-28:
I saw that post too. People are reporting that they replaced the PSU. Is that the only option?

Funny thing is that I didn't have that clicking noise before. I even don't know what I have changed ... I remember that I tuned the bed like this:

M303 E-1 S70

Recv: #define DEFAULT_bedKp 143.41
Recv: #define DEFAULT_bedKi 26.70
Recv: #define DEFAULT_bedKd 513.59

M304 P143.41 I26.70 D513.59
M500

Maybe I can try again the defaults that are in this firmware:

// Mega Zero 2 stock bed/board/firmware, 70 degrees.
#define DEFAULT_bedKp 350
#define DEFAULT_bedKi 63
#define DEFAULT_bedKd 452