kad/Marlin #46: Error adding Filament sensor to SKR E3 Turbo board

TxarlyM, 2022-04-30:
Hello. First of all congratulations for your great work with this printer. I have obtained very good results, and it is a luxury to have information like this at our disposal for those of us who do not know much about this exciting world.

I have a SKR E3 Turbo board installed in my Anycubic Mega Zero 2.0 with the latest firmware released by you, and I am very satisfied with it... The problem is that I need to add a filament sensor. I have two sensors: The first one is a standard sensor like the axis end switches, and the second one is the BigTreeTech smart sensor.

I would like to put the smart, and I would like to know if you can publish a version of this firmware with this configuration or tell me where I can download the configuration files to make the modifications myself.

Thank you for your effort publishing all this information... especially when Anycubic has this model of printer discontinued and fortunately there are several users who still support it.

Greetings from Spain

TxarlyM

ckiefer, 2022-05-06:
I am also highly appreciating the work, passion, and effort that is done on this repo -- I am using the firmware everyday. The problem with this repo is that questions are answered seldomly.

kad, 2022-07-03:
@TxarlyM hi. The default configuration for BTT boards are done with BTT Smart filament sensor in mind. In theory, it should work correctly by wiring it as in BTT documentation and then enable in Marlin menu on LCD. Mine worked well in the past.

kad, 2022-07-03:
> I am also highly appreciating the work, passion, and effort that is done on this repo -- I am using the firmware everyday. The problem with this repo is that questions are answered seldomly.

Yeah, sorry about that. This is my free-time project, and lately due to main work I don't have too much time.
Also, I got my MZ1+mods broken (hotend part), so even if I do new builds, right now I don't have anything to try it on. :(
I've ordered some spare parts, but it would take weeks to get them from China to here :(